peterhinch/micropython-async #115: Pushbutton class buggy ?

petaramesh, 2024-01-27:
Hello,

I'm trying the Pushbutton v3 class on a Raspberry Pi Pico W with micropython v1.22.1 and I'm encoutering a few issues : 

- If a Pushbutton is defined with "suppress=True" and has press_func, long_func and double_func defined, but not release_func, a short press causes a crash : 

Exception occurred !
Traceback (most recent call last):
  File "asyncio/core.py", line 1, in run_until_complete
  File "/lib/primitives/delay_ms.py", line 47, in _timer
  File "/lib/primitives/__init__.py", line 20, in launch
  File "/lib/primitives/pushbutton.py", line 83, in _ddto
AttributeError: 'Pushbutton' object has no attribute '_fa'

This doesn't occur if "suppress=True" is not used.

- Trying to define button.long_press_ms or button.double_click_ms do not produce expected results : 
- defining button.double_click_ms breaks long press time : Every single short press then also registers as a long press.
- defining button.long_press_ms doesn't seem to have any influence at all.

So basically it seems to be working good as long as no fancy options are used : No "suppress=True" and no custom delays, and then  it behaves allright.

peterhinch, 2024-01-27:
Re your first bug, I can replicate this and will fix it.

I cannot replicate your issues with the custom times. These are class variables and it is necessary to set them prior to instantiating the class. The following script works with expected times (I have defined a `release_func` to avoid triggering the above bug):
```python
from primitives import Pushbutton
from machine import Pin
import asyncio
sel = Pin(16, Pin.IN, Pin.PULL_UP)
Pushbutton.long_press_ms=2000
Pushbutton.double_click_ms=1000
pb = Pushbutton(sel, True)
def f(a):
    print(a)
pb.press_func(f, ('press',))
pb.long_func(f, ('long',))
pb.double_func(f, ('double',))
pb.release_func(f, ('release',))
async def main():
    await asyncio.sleep(10)
asyncio.run(main())
```
I will amend the docs to clarify the use of the class variables.

If you are still encountering a problem with custom times, please provide a script which reproduces the fault.

petaramesh, 2024-01-27:
Hi Peter,
Thanks for your so quick action and fix !
I'll test your code above to couble-check whether or not the timing settings actually work, but so far my attempts here tended to demonstrate they didn't.
About `suppress` not being intended to be used wihtout a release function, it were my first experiments with these routines and I was testing different methods to check what behaviour was the best for what I intended to do...
That kind of tests tend to unearth unexpected bugs ;)

Many thanks again.

petaramesh, 2024-01-27:
I tested your code with the custom timings, and it actually works.

My mistake was that I had used `mybutton.long_press_ms=2000` where it is `Pushbutton.long_press_ms=2000`

i.e. I thought it was a per-button setting, where it seems to be a global one.

peterhinch, 2024-01-27:
Glad this is all OK now.

closed 2024-01-27 by peterhinch