peterhinch/micropython-async #116: Feature request - WaitAny accepts task

sandyscott, 2024-02-01:
Hi,

I've found a few situations in my current project where I'd quite like to use a task as one of the inputs to `WaitAny`.

e.g:

```python
async def background_coro():
    # Insert work here...
    await asyncio.sleep(5)
    
async def main()
    switch = ESwitch(#some pin)
    background_task = asyncio.create_task(background_coro())
    
    await WaitAny((switch.close, background_task).wait()
    background_task.cancel()
```

I can easily solve this with passing around `Event` objects, but it muddies the picture compared to how simple waiting for the task alone would be (ie. `await background_coro()`)

I appreciate there may be lots of good reasons why this can't work, I just thought I'd share a little frustration in case there's something I've missed.

peterhinch, 2024-02-03:
I think this `ELO_maker` class will do what you want. It takes an arbitrary coroutine and converts it to an ELO. Warning: it's off the top of my head and had minimal testing. Please report back.
```python
class ELO_maker:
    def __init__(self, coro, *args, **kwargs):
        self.coro = coro
        self.args = args
        self.kwargs=kwargs

    async def wait(self):
        await self.coro(*self.args, **self.kwargs)

async def background_coro(t):  # Demo of arg passing
    # Insert work here...
    await asyncio.sleep(t)
    
async def main()
    switch = ESwitch(#some pin)
    
    await WaitAny((switch.close, ELO_maker(background_task, 5))
    background_task.cancel()
```

sandyscott, 2024-02-08:
Thank you! That's a really tidy solution.

peterhinch, 2024-02-08:
I have now pushed an update. The `ELO` class has some changes but remains very compact.

The changes were necessary because, unlike an `Event`, a `Task` can return a value. It can also be cancelled. There are no changes to `WaitAll` or `WaitAny`, and cancellation is handled in such a way that `WaitAll` and `WaitAny` behave as if the `Task` had terminated normally. The application can determine how the `Task` terminated.

It is documented and there is a [test script](https://github.com/peterhinch/micropython-async/blob/master/v3/primitives/tests/elo_test.py).

I'll close the issue as complete, but comments are of course welcome.

closed 2024-02-08 by peterhinch