peterhinch/micropython-async #117: question?

gladyscatface, 2024-03-26:
Thank you Sir, for your help in getting some asyncio code working, is there such a thing as :(sw.toggle_func)?

async def my_app():
    pin = Pin(16, Pin.IN, Pin.PULL_UP)  # Hardware: switch to gnd
    red = led
    sw = Switch(pin)
    sw.close_func(pulse, (red, 1000))  # Note how coro and args are passed
    sw.open_func(pulse, (red, 1000))
    await asyncio.sleep(60)  # Dummy application code

This was cut and pasted from your tutorial, I don't like callbacks, and  wish to build 
 mqtt switches without loops that hide away in dark places.
Thanks
Brian

peterhinch, 2024-03-26:
For a callback-free solution see [ESwitch](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/DRIVERS.md#31-eswitch-class). If you want to toggle a `Pin` each time a switch closes, adapt the sample code as follows:
```python
async def closure(pin):
    while True:
        es.close.clear()
        await es.close.wait()
        pin.value(not pin.value())
```

peterhinch, 2024-06-04:
Closed due to inactivity.

closed 2024-06-04 by peterhinch