peterhinch/micropython-async #118: queue.py: Tuples are not read back correctly

ExperiMentor, 2024-04-09:
The elements seem to be returned in the wrong order:

print(q.qsize())
0
q.put_nowait({"a", "b", "c", 42, 12})
print(q.qsize())
1
print(q.get_nowait())
{12, 'a', 42, 'c', 'b'}

peterhinch, 2024-04-10:
This has no connection with the `Queue` class. The object `{"a", "b", "c", 42, 12}` is a `set` not a `tuple`:
```py
>>> type({"a", "b", "c", 42, 12})
<class 'set'>
```
Sets are not ordered therefore they may be printed in any order.

closed 2024-04-10 by peterhinch

ExperiMentor, 2024-04-10:
Sincere apologies. My fault entirely.

I've checked, and the queue function works exactly as expected with both tuples and lists:

**Tuple (use parentheses):**
q.put_nowait(("a", "b", "c", 12, 17))
>>> x=q.get_nowait()
>>> type(x)
<class 'tuple'>
>>> print(x)
('a', 'b', 'c', 12, 17)
>>> print(x[0])
a
>>> print(x[3])
12

**List [use square brackets]**
>>> q.put_nowait(["a", "b", "c", 12, 17])
>>> x=q.get_nowait()
>>> type(x)
<class 'list'>
>>> print(x)
['a', 'b', 'c', 12, 17]
>>> print(x[0])
a
>>> print(x[3])
12