peterhinch/micropython-async #119: Import uasyncio

mishal, 2024-05-07:
Hello Peter,

first let me thank you for this great repository and docs!

In micropython v 1.22.1 (released Oct 6, 2023) uasyncio becomes asyncio (see: https://github.com/micropython/micropython/releases/tag/v1.21.0) . I do not know what version of micropython should this repo support, but imports across code in this repo imports uasyncio or uses try-catch block which tries to to import uasyncio with fallback to asyncio. The release notes comments this change as:  Now, one should just always import the non-u-version and no longer think about u-naming. 

It's not a big deal, but may cause some confusion.

peterhinch, 2024-05-08:
Thanks for the report. I have updated the tutorial text, but dealing with the various modules is on my todo list.

peterhinch, 2024-05-09:
I have now removed all references to `uasyncio` save an intentional one in the tutorial. Thanks for prompting me to do this.

closed 2024-05-09 by peterhinch