peterhinch/micropython-async #120: Question about sharing data beween async and non ansync tasks

gampam2000, 2024-05-27:
Hi, first thanks for your excellence tutorials, they helped me a lot.

I have the following challenge, I have a wireless sensor, that emits a lot of different encrypted values, for example, voltage, current, temperature, timestamp... and the decryption takes some time. So i use your ringbuffer queue for the incoming packets and then an async task to decrypt the data. This works fine so far, but I have a problem sharing this data then to different other tasks. The other tasks are a webserver, mqtt client, a Display, and other tasks that controlls some LEDs. But some of the other tasks also need to update some values in this dataset. I only need one single dataset. 

I created a singelton Class with a dictionary in it that uses the asyncio.lock. With async and sync_get and update methods, but on my ESP32-S3 this crashes the Micropython OS (1.22.1):

This is a reduced test program that crashes:

```
import asyncio
import random
import sys

class SensorData:
    _instance = None

    def __new__(cls, *args, **kwargs):
        if cls._instance is None:
            cls._instance = super().__new__(cls, *args, **kwargs)
        return cls._instance

    def __init__(self):
        if not hasattr(self, 'initialized'):
            self.data = {
                "timestamp": None,
                "unixts": 0,
                "decryption_key_configured": False,
                "decryption_successful": False,
                "l1_voltage": {"value": 0, "unit": "V"},
                "l2_voltage": {"value": 0, "unit": "V"},
                "l3_voltage": {"value": 0, "unit": "V"},
                "l1_current": {"value": 0, "unit": "A"},
                "l2_current": {"value": 0, "unit": "A"},
                "l3_current": {"value": 0, "unit": "A"},
                "wifi_mac": None,
                "wifi_ip": None,
                "wifi_rssi": None
            }
            self.lock = asyncio.Lock()
            self.initialized = True

    async def update(self, keys, value):
        async with self.lock:
            print(f"update lock {keys} {value}")
            await asyncio.sleep(0)
            return await self._update(keys, value)

    async def _update(self, keys, value):
        d = self.data
        for key in keys[:-1]:
            d = d[key]
        d[keys[-1]] = value
        return value
    
    #blocks until the lock
    def sync_update(self, keys, value):
        print(f"sync_update {keys} {value}")
        return asyncio.run(self.update(keys, value))

    async def get(self, keys=None):
        async with self.lock:
            print(f"get lock {keys}")
            await asyncio.sleep(0)
            return await self._get(keys)
    
    async def _get(self, keys=None):
        if keys is None:
            return self.data.copy()
        d = self.data
        for key in keys:
            d = d[key]
        return d

    #blocks until the lock
    def sync_get(self, keys=None):
        print(f"sync_get {keys}")
        return asyncio.run(self.get(keys))    

    def print_value(self, keys, round_digits=None):
        try:
            data = asyncio.run(self.get(keys))
            value = data["value"]
            unit = data.get("unit", "")
            unitprefix = ""

            if value >= 1_000_000_000:
                value /= 1_000_000_000
                unitprefix = "G"
            elif value >= 1_000_000:
                value /= 1_000_000
                unitprefix = "M"
            elif value >= 1_000:
                value /= 1_000
                unitprefix = "k"

            if round_digits is not None:
                value = round(value, round_digits)

            return f"{value} {unitprefix}{unit}"

        except Exception as e:
            print(f"keys: {keys}")
            sys.print_exception(e)

async def display():
    md = SensorData()  # Singleton instance
    while True:
        l1 = await md.get(["l1_voltage","value"])
        l2 = await md.get(["l3_current","value"])
        print("Async Data: {} {}".format(l1, l2))

        print(md.print_value(["l1_voltage"]))
        print(md.print_value(["l2_voltage"]))
        print(md.print_value(["l3_voltage"]))
        print(md.print_value(["l1_current"], 3))
        print(md.print_value(["l2_current"], 3))
        print(md.print_value(["l3_current"], 3))

        await asyncio.sleep(5)

async def update():
    md = SensorData() # Singleton instance
    while True:
        v1 = random.randrange(0,500,1)
        i1 = random.randrange(0,5000_000,100)
        print(f"random: {v1} {i1}")
        await md.update(["l1_voltage", "value"], v1)
        await md.update(["l2_voltage", "value"], 1_111)
        await md.update(["l3_voltage", "value"], 2_111_112)
        await md.update(["l1_current", "value"], 3_111_111_113)
        md.sync_update(["l2_current", "value"], i1) # this crashes the OS
        await asyncio.sleep(3)

async def main():
    asyncio.create_task(display())
    asyncio.create_task(update())
    await asyncio.sleep(0)

md = SensorData()   
md.sync_update(["l3_current", "value"], 6_200)
print(md.print_value(["l3_current"], 3))

loop = asyncio.get_event_loop()
loop.create_task(main())
loop.run_forever()
print("i should never get printed")
```

This crashes with this error:

```
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x29 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382b8e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x105c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbd8
load:0x403cc700,len:0x2e30
entry 0x403c989c
sync_update ['l3_current', 'value'] 6200
update lock ['l3_current', 'value'] 6200
get lock ['l3_current']
6.2 kA
get lock ['l1_voltage', 'value']
random: 227 2411200
update lock ['l1_voltage', 'value'] 227
get lock ['l3_current', 'value']
Async Data: 0 6200
update lock ['l2_voltage', 'value'] 1111
get lock ['l1_voltage']
227 V
update lock ['l3_voltage', 'value'] 2111112
get lock ['l2_voltage']
1.111 kV
update lock ['l1_current', 'value'] 3111111113
sync_update ['l2_current', 'value'] 2411200
get lock ['l3_voltage']
update lock ['l2_current', 'value'] 2411200
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42032149  PS      : 0x00060b30  A0      : 0x8202747e  A1      : 0x3fcbcc10  
A2      : 0x00000003  A3      : 0x3fcbcd68  A4      : 0x00040006  A5      : 0x3c1b31ac  
A6      : 0x3c1b31b0  A7      : 0x3c1b1e00  A8      : 0x00000006  A9      : 0x3fcbcdd0  
A10     : 0x3c1b1a40  A11     : 0x00000006  A12     : 0x00000002  A13     : 0x3fcbce5c  
A14     : 0x00000180  A15     : 0x3c1b1b20  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0000001a  LBEG    : 0x4204d74a  LEND    : 0x4204d753  LCOUNT  : 0x00000000  

Backtrace: 0x42032146:0x3fcbcc10 0x4202747b:0x3fcbcc40 0x4202df9d:0x3fcbcc70 0x4202e065:0x3fcbcc90 0x40378e2a:0x3fcbccb0 0x42027413:0x3fcbcd50 0x4202df9d:0x3fcbcdb0 0x4202e233:0x3fcbcdd0 0x40379019:0x3fcbce10 0x4202775f:0x3fcbceb0 0x42027841:0x3fcbced0 0x4202730a:0x3fcbcf00 0x4202df9d:0x3fcbcf20 0x4202e065:0x3fcbcf40 0x40378e2a:0x3fcbcf60 0x42027413:0x3fcbd000 0x4202df9d:0x3fcbd030 0x40378d87:0x3fcbd050 0x42027413:0x3fcbd0f0 0x4202df9d:0x3fcbd150 0x4202e065:0x3fcbd170 0x40378e2a:0x3fcbd190 0x42027413:0x3fcbd230 0x4202df9d:0x3fcbd260 0x4202e065:0x3fcbd280 0x40378e2a:0x3fcbd2a0 0x4202775f:0x3fcbd340 0x42027841:0x3fcbd360 0x4202730a:0x3fcbd390 0x4202df9d:0x3fcbd3b0 0x4202e065:0x3fcbd3d0 0x40378e2a:0x3fcbd3f0 0x42027413:0x3fcbd490 0x4202df9d:0x3fcbd4c0 0x40378d87:0x3fcbd4e0 0x42027413:0x3fcbd580 0x4202df9d:0x3fcbd5d0 0x4202e065:0x3fcbd5f0 0x40378e2a:0x3fcbd610 0x42027413:0x3fcbd6b0 0x4202df9d:0x3fcbd710 0x4202e00a:0x3fcbd730 0x4204e359:0x3fcbd770 0x4204e4e6:0x3fcbd7a0 0x4202e0e4:0x3fcbd890 0x403790b7:0x3fcbd8d0 0x42027413:0x3fcbd970 0x4202df9d:0x3fcbd9c0 0x4202dfb2:0x3fcbd9e0 0x4203a453:0x3fcbda00 0x4203a7ab:0x3fcbda90 0x4204346e:0x3fcbdac0

ELF file SHA256: b35dce624

Rebooting...
```

Am I doing something wrong? Or is this a micropython bug?
Any suggestions what would be a good and safe approach to share data beween different tasks? 

Thanks & best regards

peterhinch, 2024-05-28:
I can see one obvious problem.
```py
    def sync_get(self, keys=None):
        print(f"sync_get {keys}")
        return asyncio.run(self.get(keys))    
```
The function `asyncio.run` should be used once only to start `asyncio`. I think the method should read
```py
    def sync_get(self, keys=None):
        print(f"sync_get {keys}")
        return await(self.get(keys))    
```
In general it's a lot to ask for someone to debug a substantial body of code. If you think you've found a bug you should systematically reduce your code until you have a minimal "reproducer" that a maintainer can run.

gampam2000, 2024-05-30:
Hi,

i can't change it this way because if i want to use the `await` the function needs to be an async function. But i want to be able to call it from a non async context. Is this even possible?

Do you have any suggestion on how to update data in a dictionary from async and non async tasks? My idea was to use asyncio.lock.

I have stripped down my testcode but so far i could not figure out why or where exactly it crashes, but i quess you are correct that it happens because asyncio.run is run multiple times.

Thanks & best regards

peterhinch, 2024-05-31:
> i can't change it this way because if i want to use the await the function needs to be an async function. 

Good point.

In general I'm puzzled by your code. There doesn't seem to be a clear distinction between synchronous and asynchronous code. For example the following is declared as `async` but is in fact synchronous - it never yields to the scheduler. 
```py
    async def _get(self, keys=None):
        if keys is None:
            return self.data.copy()
        d = self.data
        for key in keys:
            d = d[key]
        return d
```
This function is inherently synchronous. It could be made asynchronous by adding `await asyncio.sleep(0)` statements but there would be no practical benefit. This is because I would expect it to execute very quickly - probably in a very few ms or less. There is no practical gain in yielding to the scheduler unless a routine is likely to block for high tens of ms or longer.

The only point in yielding is to enable other tasks to run. Say you had a routine that, implemented as synchronous code, might block for (say) one second. You might well decide that having other tasks stall for 1s is unacceptable. So you change the routine to an async one, splitting the task into ten segments, with `await asyncio.sleep(0)` after each segment. Then the blocking time is reduced to 100ms.

My approach would be to review the code in this light, writing as async only those routines that threaten to harm the responsiveness of other async tasks.

To answer your question on running an async task from a synchronous function, you are correct in saying this is impossible. You can launch it:
```py
def foo():  # Synchronous
    task = asyncio.create_task(bar())
    print('got here')
```
but the print statement is executed immediately. You wouldn't want the synchronous function to block while `bar()` was running because a blocking synchronous function locks up the scheduler.

In such cases there are various ways to check whether `bar()` has completed. It might run a callback or trigger an `Event` for example.

peterhinch, 2024-05-31:
To add to the above, I suspect that you are unclear about the way `asyncio` works. Consider this function, which reads 1 million bytes from a binary file and returns their sum.
```py
def add(file_path):
    with open(file_path, 'rb') as f:
        result = 0
        for _ in range(1_000_000):
            result += int(f.read(1))
    return result
```
The execution time is heavily dependent on the hardware hosting the file, but let's assume it takes 5 seconds. Let's also assume that you want other tasks to run while this is in progress. A naive user might do this:
```py
async def add(file_path):
    with open(file_path, 'rb') as f:
        result = 0
        for _ in range(1_000_000):
            result += int(f.read(1))
    return result
```
with another task calling this as follows:
```py
async def foo():
    total = await add("my_file")
```
This would run, but the scheduler would lock up for 5s. No other work would be accomplished while the file read was in progress. `async def` simply tells the compiler that the coroutine might include `await` statements. As written, there are none, so the routine would block until it completed. It is still a synchronous function with the annoying feature that it can't be used with a simple function call.

To actually achieve concurrency you need to do something like:
```py
async def add(file_path):
    with open(file_path, 'rb') as f:
        result = 0
        for _ in range(100):
            for _ in range(10_000):
                result += int(f.read(1))
            await asyncio.sleep(0)  # Other tasks get some runtime here
    return result
```
This is now a true coroutine. Other tasks get some execution time about every 50ms (5s/100). The calling task
```py
async def foo():
    total = await add("my_file")
```
sees a 5s pause before it gets the total - this is inevitable - but other tasks continue to run.

I hope this provides some clarity.

To add to this, a dictionary access is so fast that it isn't worth attempting to implement as asynchronous code.

peterhinch, 2024-06-05:
Asyncio is a cooperative scheduler. A task only gets to run when another has yielded control to the scheduler. This means that synchronous code is never pre-empted. This includes synchronous code running as part of a coroutine. This is a major benefit in that locks are not required to protect synchronous code - see [this tutorial section](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md#85-why-cooperative-rather-than-pre-emptive).

If multiple tasks are accessing a `dict` nothing will break (even if tasks are updating the `dict`). A lock might be required if it's important that several asynchronous updates must be read as a coherent set.
```py
async def dict_update(d, lock):
    while True:
        async with lock:
            d['x'] = await read_sensor_x()   # Function yields to scheduler while waiting for a device
            d['y'] = await read_sensor_y()  # ditto
        await asyncio.sleep(1)
```
A task using those `dict` keys will get a consistent pair of values if it protects the accesses with the lock. Without protection it might get a new `x` value with a prior `y` value. Whether this matters, and whether a lock is required, is application dependent.

gampam2000, 2024-06-09:
Hi, thanks for your explanation. Now I understand when to use a lock or not. 
Best regards!

closed 2024-06-09 by gampam2000