peterhinch/micropython-async #130: I2C address in HTU21D constructor

runout-at, 2024-12-19:
Thanks for your great work!

I have a wish for your HTU21D driver. `htu21d_mc.py`

Could the I2C address be a parameter in the `__init__` function?

```
def __init__(self, i2c, address=_ADDRESS, read_delay=10):
    self.address = address
```

and `self.address` be used subsequently.

The background for this request is an I2C class  which extends the I2C class from micropython and can handle I2C multiplexers like TCA9548A by adding 12bit to the I2C address. It works with `machine.I2C` and `machine.SoftI2C` too.
The code is backward compatible.

The idea behind it from my docu:
```
    changes to machine.I2C
    if addr > 0xff assume the address has following hex-format:
    8 bit multiplexer address, 4 bit channel, 8 bit target address
    
    eg 0x70140 for a htu21 sensor with address 0x40
    on channel 0 of a TCA9548A multiplexer with address 0x70
    
    channels are 1..8; corresponding to 0..7 on the TCA9548A
    channel 0 deselects all channels
    
    extract the address/channel by bit-shifting and and-operation.
    select the multiplexer channel
    deselect channel after read/write
```

The code is at https://gitlab.com/runout/micropython/-/blob/main/i2c.py

peterhinch, 2024-12-20:
I have pushed this change. Please could you test and report back (I don't have an HTU21D).

Note I use the third arg for address - this avoids breaking existing code where `read_delay` is used.

runout-at, 2024-12-20:
wow. that was fast :)
THX! I will test in the next days.

With my extended I2C class for using a TCA9548A I2C-multiplexer I use addresses like 0x70140.
Where
70 is the address of the multiplexer,
1 is the channel to select on the multiplexer
40 is the address of the HTU21.

Link to the code I did include in my first post.

peterhinch, 2024-12-21:
I understand this, and for a while I was convinced. Then doubts crept in...

In hardware terms, I assume that the TCA9548A strips off the 12 bits so that the bus to which the HTU21D sees 0x40. I haven't studied the TCA9548A in detail, but it strikes me that it must do this otherwise attached devices would malfunction.

I don't have any problem with having the option to specify the address: this can enable a driver to support Chinese clone chips with a different hardware address. But in your case I think you'll need to accept the default.

runout-at, 2024-12-22:
The 12 bits will be stripped by my extended I2C class and after selecting the channel just the 0x40 will be used to address the HTU21D.

Its the method selChannel() around line 39-55 in https://gitlab.com/runout/micropython/-/blob/main/i2c.py
which will return the 0x40 to the I2C-read/write functions.

runout-at, 2024-12-22:
The code mostly works.

The only problem is in

```
        if address not in self.i2c.scan():
            raise OSError("No HTU21D device found.")
```

as the scan function does not know on which TCA9548A and on which channel to search for the HTU21D.

If i remove the 2 lines it works perfectly.

For my I2C class I do override the original scan function and I can provide an address with channel, eg `i2c.scan(0x70100) `. But using this in your code would break backward compatibility.

It looks like I have to maintain my own version of your HTU21D driver anyways.

Thanks!

peterhinch, 2024-12-23:
OK, I think we'll have to leave it there.

closed 2024-12-23 by peterhinch