peterhinch/micropython-async #131: sdcard

omeErik, 2024-12-24:
Hi Peter: thanks a lot for your documents, I like and used them a lot to get going with async. For my project, I'm using an sdcard library, and it is blocking, for larger files up to seconds. The driver itself is not that complicated, making it await in the loops can be done easily. However, how it works in the grander scheme of things is not clear at all to me: since you cannot start it as a task (the card is enabled by a mount), there is no way for me to make it async. Allow me to ask if you are aware of a way out. Thanks a lot! - Erik

peterhinch, 2024-12-26:
You can't avoid the fact that a file read blocks, but you can read the file in a piecemeal fashion so that blocking periods are limited. See [this example](https://github.com/peterhinch/micropython-micro-gui/blob/7edc54e54aa54f971afd2d6ded79ab01b4c5928a/gui/demos/audio.py#L232) where an audio file is read concurrently with playing the file via I2S and refreshing a screen. The refresh also blocks but with careful design aimed at ensuring that nothing blocks for longer than a threshold continuous play can be achieved.

omeErik, 2024-12-26:
Excellent, thanks, I see. It never crossed my mind, but the way is blocks, of course it is. Precisely the tip I was looking for. With this I should be able to figure it out. Thanks again - Erik

omeErik, 2024-12-26:
done.

closed 2024-12-26 by omeErik