peterhinch/micropython-async #134: Design advice - please

adrianblakey, 2025-02-13:
Please could you suggest how I could arrange to simultaneously write the same ADC values to flash, a websocket and BT?

My device hardware provides data on 3 ADC inputs to a Pi Pico 2 W, I read them “simultaneously”, round robin using DMA into a single element blocking ThreadSafeQueue (TSQ) on core 1 using _threads. (Code below). I decided to offload the mcu by doing this in hardware and not using the AADC class.

I want to write the three collected values as a timestamped tuple about once every microsecond, to none, one, or all of: a flash memory file, a websocket, bt notification service. File writes depend on a button toggle; ws writes on whether or not I have a WiFi connection (and microdot web server listening), and BT notifications on whether or not I have BT client connected.

I think I need a broker that runs continuously on core 0 to consume the TSQ, and unblocks it so it iteratively collects another set of ADC values. I want to be able to dynamically subscribe/unsubscribe the 0-3 “sinks” to the broker. The sinks might be FiFO’s for the file/ws/bt. When the broker runs should it write the tuple from the TSQ into one or more FIFO’s? If the there are no subscribed (or enabled?) FIFO’s it should just return.

There shall be a flash/sdcard writer task (coded and tested), a button task that toggles on/off file collection. When it’s turned on, the file writer could (re)subscribe (or be enabled?) to the broker and start consuming its FIFO.

Similarly if a WiFi connection is successful, a Microdot server is started with a Websocket (@with_websocket) method, which once inititalized subscribes to the broker, runs about every 2 milliseconds to read from its FIFO and writes to the browser. The webserver code is tested and runs.

And - there’s an async bluetooth notification task that needs to do something similar - the code for this is tested.

A few questions:

1. Would it be best to use event based programming to do this? (Note: I use Pushbutton to process the button to stop/start file data collection) Should (can I?)I write the ADC tuple to the sinks as soon as it appears (if they are open) and forget about the TSQ?
2. I want to use the second core to process the ADC data and use the first core for the UI etc. Should I use the Context class to encapsulate the _thread code that processes the ADC’s? 
3. Should I use the Broker class to read the TSQ and provide the value to the file/ws/bt subscribers  - one in, many out, one topic?
4. I currently do all the file writes synchronously - I can’t find any way to do this asynchronously - is that correct? These might be slow and block the broker or overflow a queue.
5. Should I use the Agent class to encapsulate the subscriptions?
6. Should I use the RingbufQueue (in a Broker) to communicate the ADC tuple (read from the TSQ) to the sinks?

The ADC code is like this (not ready yet to make my repo public - but if necessary I will):
```

import asyncio
from threadsafe import ThreadSafeQueue
import _thread

import time, array, uctypes, rp_devices as devs
import logging
from micropython import const

log = logging.getLogger("adc-devs”)
# 12 bit adc 0-4095
# Only 9.3 bits are useful

IF: float = const(3.3 / 4096)    
VC: float = const(17.966 / 4096) 
zero_current: int = 2048      

start_ticks: int = const(time.ticks_us())  # Start time

ADC_CHAN_0 = const(0)
ADC_CHAN_1 = const(1)
ADC_CHAN_2 = const(2)

ADC_PIN_26  = const(26)
ADC_PIN_27  = const(27)
ADC_PIN_28  = const(28)

ADC_SAMPLES = 3
MIN_SAMPLES, MAX_SAMPLES = 1, 10
ADC_RATE = 100000
MIN_RATE, MAX_RATE = 1000, 500000
DMA_CHAN = 0

interval_ms: int = const(1)    # ADC Collection interval millisec

adc = devs.ADC_DEVICE
dma_chan = devs.DMA_CHANS[DMA_CHAN]
dma = devs.DMA_DEVICE

# Init the ADC pins
def gpio_init():
    pin_26 = devs.GPIO_PINS[ADC_PIN_26]
    pin_27 = devs.GPIO_PINS[ADC_PIN_27]
    pin_28 = devs.GPIO_PINS[ADC_PIN_28]
    pad_26 = devs.PAD_PINS[ADC_PIN_26]
    pad_27 = devs.PAD_PINS[ADC_PIN_27]
    pad_28 = devs.PAD_PINS[ADC_PIN_28]
    pin_26.GPIO_CTRL_REG = pin_27.GPIO_CTRL_REG = pin_28.GPIO_CTRL_REG = devs.GPIO_FUNC_NULL
    pad_26.PAD_REG = pad_27.PAD_REG = pad_28.PAD_REG = 0
    pad_26.PAD.IE = pad_27.PAD.IE = pad_28.PAD.IE = 0x0 
    pad_26.PAD.OD = pad_27.PAD.OD = pad_28.PAD.OD = 0x1
 
# Initialise ADC DMA & FIFO to read ADC 0 - 2 roundrobin
def adc_dma_init(): 
    dma.CHAN_ABORT = 0xffff

    adc.FCS.SHIFT = 0                   # No 8 bit shift in FIFO
    adc.FCS.EN = adc.FCS.DREQ_EN = 1
    adc.FCS.THRESH = adc.FCS.OVER = adc.FCS.UNDER = 1
    adc.CS.RROBIN = 0x07                # Mask, set bits 0, 1, 2 for channels 0, 1, 2

    dma_chan.READ_ADDR_REG = devs.ADC_FIFO_ADDR
    dma_chan.CTRL_TRIG_REG = 0
    dma_chan.CTRL_TRIG.CHAIN_TO = DMA_CHAN
    dma_chan.CTRL_TRIG.INCR_READ = dma_chan.CTRL_TRIG.INCR_WRITE = dma_chan.CTRL_TRIG.IRQ_QUIET = 1
    dma_chan.CTRL_TRIG.TREQ_SEL = devs.DREQ_ADC
    dma_chan.CTRL_TRIG.DATA_SIZE = 0x1  # Half word transfers
    dma_chan.CTRL_TRIG.EN = 0
    
# Discard any data in ADC FIFO
def flush_adc_fifo():
    while adc.FCS.LEVEL:
        x = adc.FIFO_REG
        
# Single ADC 0 capture not using DMA - not used
def capture_once(enob: bool = False):
    # Call adc_init first
    adc.CS.START_ONCE = 1             # Read ch 0 once
    print('12 bit int Result:', adc.RESULT.VALUE)
    print('Result register bits: {:012b}'.format(adc.RESULT.VALUE))
    if enob:
        adc.RESULT.VALUE >>= 3  # Clean up, only low 9 bits are useful ENOB 9.2
        print('Result register ENOB bits: {:09b}'.format(adc.RESULT.VALUE))
    adc.CS.START_ONCE = 0
    return adc.RESULT.VALUE

# Measure the zero current on ADC 0, 16 times and average
def calibrate_current() -> None:
    count: int = 0
    for i in range(16): 				# get it 16 times and shift >> 4 to divide
        count += capture_once()
        time.sleep_ms(interval_ms)
    zero_current = count >> 4

# Capture ADC 0-2 samples using DMA
def adc_capture(nsamples: int = ADC_SAMPLES, xrate: int = 100000):
    flush_adc_fifo()
    nsamp = max(MIN_SAMPLES, min(MAX_SAMPLES, nsamples))
    adc_buff = array.array('H', (0 for _ in range(nsamp)))
    if xrate == 0:
        adc.DIV.INT = 0         # Min interval
    else:
        rate = max(MIN_RATE, min(MAX_RATE, xrate))
        adc.DIV_REG = (48000000 // rate - 1) << 8
    adc.CS.RROBIN = 0x07        # Grab 0, 1, 2
    dma_chan.WRITE_ADDR_REG = uctypes.addressof(adc_buff)
    dma_chan.TRANS_COUNT.COUNT = nsamp
    dma_chan.TRANS_COUNT.MODE = 0x0 # Decrement the count, trigger myself
    dma_chan.CTRL_TRIG.EN = 1
    adc.CS.START_MANY = 1
    while dma_chan.CTRL_TRIG.BUSY:
        time.sleep_us(2)
    adc.CS.START_MANY = 0
    dma_chan.CTRL_TRIG.EN = 0
    return adc_buff

# Write adc's to queue 
def producer(a_queue):
    while True:
        buff = adc_capture(3, 0)
        tick = time.ticks_diff(time.ticks_us(), start_ticks)
        v = f"{tick},{buff[2] * VC:1.3f},{buff[1] * VC:1.3f},{(buff[0] - zero_current) * IF / .025:1.3f}”
        #print(v)
        #v = ",".join([("%1.3f" % (val*3.3/4096)) for val in adc_capture(3, 0)])
        #print(v)
        a_queue.put_sync(v, block=True)  # Wait if queue fills.
        #time.sleep_ms(interval_ms)
        
def rundevice(tsq):
    calibrate_current()
    adc_dma_init()
    _thread.start_new_thread(producer, (tsq,))    
    
gpio_init()
adc_init()
TSQ = ThreadSafeQueue(1)  

# TODO run a consumer test so this doesn't just block
if __name__ == “__main__”:
    rundevice(TSQ)             # This runs on core 2

```

peterhinch, 2025-02-13:
Properly analysing this is a nontrivial task requiring a good deal of extra information on things like peak and average data rate, data volume and latency limitations. You'd also need to define exactly which processes need to run concurrently. In a commercial setting I'd expect to sit down with a couple of subject specialists for a day or two to get to produce an outline design. At the moment I'm unsure whether this project is even suitable for `asyncio`.

I can tell you that writing to Flash or SD card blocks for a variable period. A flash segment erase can take tens of ms, and if you are writing to a file you have no means of knowing when this will occur. I have no experience with BT nor with high performance uses of HTTP/HTTPS.

To make a start you need to characterise each interface to determine its blocking behaviour, throughput capability, and latency. You might do this by measurement or by asking for assistance elsewhere.

adrianblakey, 2025-02-13:
Thanks for your reply.