peterhinch/micropython-async #135: Feedback

K0I05, 2025-02-22:
This is a great library, thanks for sharing.  

The schedule module is very handy and a decent compromise within the MicroPython environment.  I made some modifications to the schedule module, source is here: [VS Code + PyMakr](https://github.com/K0I05/XIAO-ESP32S3-PyMakr/tree/main).  

I created an esp-idf component called time-into-interval that is kind of similar, I may port this to MicroPython.  The time_into_interval function returns true when the interval condition has elapsed and it is synchronized with the clock.  I use this within a task when you need to trigger an action at a specific interval.

`
time_into_interval_config_t dt_1min_tii_5min_cfg = {
        .interval_type      = TIME_INTO_INTERVAL_SEC,
        .interval_period    = 5 * 60,
        .interval_offset    = 10
    };

// create a new time-into-interval handle - task system clock synchronization
time_into_interval_new(&dt_1min_tii_5min_cfg, &dt_1min_tii_5min_hdl);
if (dt_1min_tii_5min_hdl == NULL) ESP_LOGE(APP_TAG, "time_into_interval_new, new time-into-interval handle failed");

for ( ;; ) {
        /* print 'hello' every 5-minutes (i.e. 12:00:10, 12:05:10, 12:10:10, etc.) */
        if(time_into_interval(dt_1min_tii_5min_hdl)) {
            ESP_LOGI(APP_TAG, "Hello World");
        }
 }
`

[esp-idf time-into-interval component](https://github.com/K0I05/ESP32-S3_ESP-IDF_COMPONENTS/blob/main/components/schedule/esp_time_into_interval/README.md)

K0I05, 2025-02-22:
> This is a great library, thanks for sharing.
>
> The schedule module is very handy and a decent compromise within the MicroPython environment. I made some modifications to the schedule module, source [...]

I was able to port the time-into-interval c code to MicroPython: [time-into-interval](https://github.com/K0I05/XIAO-ESP32S3-PyMakr/blob/main/scheduler/tii.py)

An example is available here: [example](https://github.com/K0I05/XIAO-ESP32S3-PyMakr/blob/main/main.py)

peterhinch, 2025-06-16:
While this is interesting I don't think there is a valid bug report here, so closing.

closed 2025-06-16 by peterhinch