peterhinch/micropython-async #142: Documentation request threadsafe.Message.__iter__()

gvdl, 2025-06-24:
```py
    def __iter__(self):
        yield from self.wait()
        return self._data
```

I've been trying to work out what this does by reading your tutorial and Message doc. With my very limited experience using asyncio on micropython, this seems to be illegal. You `self.wait()` is declared as a coro and I thought it was illegal for a non-async function to call an async coro.

In this case you are using `yield from` to coro. Does is this bridge the asyncio divide into a syncronous iterator? If so could can I add a `"""` function docstring to explain what is going on to other confused people like me?

gvdl, 2025-06-24:
Huh found the reference, as usual your docs are amazingly detailed. The section that explains what is happeing is in Section 4.1 Awaitable classes in TUTORIAL.md. Would you mind if I add this as a link in a function docstring?

peterhinch, 2025-06-25:
I don't see a need for this: the code examples in this repo are intended to be studied in conjunction with the tutorial. I would assume that a user's first port of call when needing an clarification of a feature would be to look there.

However you score 10/10 for spotting the unusual coding :)